kriswiner/MPU6050 #11: Self test

blschreiner, 2016-01-18:
The self test implementation doesn't make sense to me, and doesn't appear to match the guidance in the register map (rev 4.2 at least).  Is there some additional documentation from Invensense supporting your implementation?  The expression at line 692 effectively calculates the percent difference between the (encoded) values stored in registers 0x0d-0x10, and the (decoded) values produced using the conversion expressions in lines 676-681.  For a given chip this result will be fixed and will be pretty close to -100%.  How does this test the sensor?  Isn't the factory trim result supposed to be compared instead with Self Test Response, which the register map defines as the difference between a (real-time) sensor reading with self-test enabled and a (real-time) sensor reading with self-test disabled?  If this understanding is wrong I'd be interested in knowing that.

kriswiner, 2016-01-18:
It is wrong.

See attached.

<https://github.com/kriswiner/MPU-6050/issues/11> .
<https://github.com/notifications/beacon/AGY1qnF_Da-e6Kuf6cdlV0ddI7iPd9Umks5
pbNjhgaJpZM4HG55W.gif>

kriswiner, 2016-01-19:
Did you get the attachment? If not PM me at onehorse@earthlink.net for the Invensense self test AN.

gahcep, 2017-12-12:
Hi @kriswiner , the email seems not working. Could you please provide a valid one?

kriswiner, 2017-12-12:
What is your question?

gahcep, 2017-12-12:
Actually the same as previous one - about the self calibration procedure. Does the Motion Driver archive from InvenSense has the self-test procedure well described? If not, can I have the self-test description document?

kriswiner, 2017-12-12:
Send me an e-mail at tleracorp@gmail.com

rossihwang, 2020-02-11:
Would you also provide me a self test AN?

kriswiner, 2020-02-11:
In the documents folder of the github repository

rossihwang, 2020-02-11:
@kriswiner, thanks, but do you mean the one in MPU9250 repo?

kriswiner, 2020-02-11:
This is defined in the PS for the MPU6050.

rossihwang, 2020-02-11:
@kriswiner I still don't have any clue 
1. The implement doesn't read the "Gyroscope Output with Self-Test Disable"
2. Why it needs add 100 here
` for (int i = 0; i < 6; i++) {
     destination[i] = 100.0 + 100.0*((float)selfTest[i] - factoryTrim[i])/factoryTrim[i]; // Report percent differences
   }`
Thanks

kriswiner, 2020-02-11:
Percent difference.

Read the Product spec and make your own self test function.

VitoLeo92, 2020-10-19:
Sorry,
I have a doubt. If the Self test> 1.0f the MPU is broken? or it can be solved with calibration?

kriswiner, 2020-10-19:
Self test is simply to check that the MPU9250 is still functioning the same
as it was at the factory. There is no calibration for "improving" the self
test. It is just a diagnostic.

The self test only uses the first 8 bytes of the result, and in some of the
sketches this is not done properly. So you might get a different result if
you just look at the last eight bytes, etc i.e., do the self test properly,
etc.