kriswiner/MPU6050 #12: 1kHz IMU for Google Cardboard ?

motorsep, 2016-01-23:
Do you happen to know of any plug and play IMU kit that would work with (almost) any Android phone for Google Cardboard head tracking? (to match whatever sensors Gear VR has)

kriswiner, 2016-01-23:
Not sure what you mean by plug and play. If you want absolute orientation in
3D with high accuracy I would recommend the EM7180 plus MPU9250 combo as
described here: 

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/

I requires an I2C-compatible microcontroller to read the data; it doesn't
just plug into the smartphone and give you head tracking info. It would take
a bit of work to do the interfacing.

I don't know if this answers your question since I don't really know what
you are asking...

<https://github.com/kriswiner/MPU-6050/issues/12> .
<https://github.com/notifications/beacon/AGY1qo2lF3NabcgyRTYnzokacrK3cT4Hks5
pctpegaJpZM4HK1Ux.gif>

motorsep, 2016-01-23:
By plug and play I mean something like this project: http://open-gear.com/ - a board with IMU that plus into USB port of the smartphone and I am guessing devs interfaced it with Oculus SDK to get it working.

That's what I am looking for :)

If there is such board, and it's accessible to masses price wise, maybe Google can implement support for it in their Cardboard SDK.

kriswiner, 2016-01-23:
Well, there is this device which has an absolute orientation engine, a K20
Kinetis processor, and can connect to a USB-enabled device:

I just designed it for a different purpose but might serve yours.

<https://github.com/kriswiner/MPU-6050/issues/12#issuecomment-174126795> .
<https://github.com/notifications/beacon/AGY1quAd2FYQ5ka618yQ9RSETSfBDvZtks5
pcvF-gaJpZM4HK1Ux.gif>

kriswiner, 2016-01-23:
Forgot the link:

https://www.oshpark.com/shared_projects/93nW3o4E

<https://github.com/kriswiner/MPU-6050/issues/12#issuecomment-174126795> .
<https://github.com/notifications/beacon/AGY1quAd2FYQ5ka618yQ9RSETSfBDvZtks5
pcvF-gaJpZM4HK1Ux.gif>