kriswiner/MPU6050 #13: doing fusion oupdates with no new data

manitou48, 2016-04-06:
Kris, I was hoping you would respond to this teensy forum

https://forum.pjrc.com/threads/33328-Prop-Shield-Beta-Test?p=101499&viewfull=1#post101499

Your prop-shield sketch does fusion updates even if there is no new data, and you argue in your MPU-6050 discussion for using a faster MCU to do additional filter updates.  Paul thinks otherwise, and I'm curious...

kriswiner, 2016-04-06:
I left a detailed response; I don't know if Paul is referring specifically to the Madgwick filter. If so, I think he is mistaken...

welltechn, 2019-02-18:
Hello Cris

Thanks for all the info you share! There are several threads discussing the topic of multiple iteration of Madgwick filter but still need clarification on one point.
You are saying that a rule of thumb is to use Madgwick iteration frequency as about 5 times of IMU sampling rate. So if IMU sampling rate is 200Hz need to run Madgwick at 1kHz. But what should be sampling frequency parameter at Madgwick at this case?
For example, I use Data ready interrupt at 6050 and get data from 6050 every 5ms (Sample_Freq = 200Hz). After I read all data from 6050 I am ready to run Madgwick. But if I want to do lets say N iterations do I need to use the same "fresh" portion of data and (Sample_Freq * N) at "Integrate rate of change of quaternion to yield quaternion" or?

Thanks in advance!

kriswiner, 2019-02-18:
I am not sure I understand your question.

For each new data (gyro) sample you want to iterate the Madgwick function 5
- 10 times, so

if(interrupt) {
read_data();

for ( i = 0; i < 10 ; i++)
{
Madgwick_call)();
update deltat;
}

} // end interrupt handling

The deltat is the time between Madgwick calls.

See here
<https://github.com/kriswiner/Ladybug/blob/master/MPU9250_MS5637_BasicAHRS2_Ladybug.ino>
for an example.

Note that this isn't exactly right since deltat is on the order of the
interrupt time for the first Madgwick call and very much smaller for the
subsequent calls, but this works because the first call allows the
quaternion to jump to the new value consistent with the new data sample and
the subsequent calls allow the solution to iterate to an asymptotic
solution.

welltechn, 2019-02-19:
Thanks a lot! Now is clear. Intuitively such iterations using the same input data feels weird but math is a fun thing, would be interesting to model that and see if the result is really approaching asymptoticaly some specific value.