kriswiner/MPU6050 #14: About the Soft-Iron Calibration

ijunglee, 2016-08-25:
Hello, I have read the hard-iron and soft-iron magnetic field calibration in your Wiki 'Simple and Effective Magnetometer Calibration':
`// Get hard iron correction
 mag_bias[0]  = (mag_max[0] + mag_min[0])/2;  // get average x mag bias in counts
 mag_bias[1]  = (mag_max[1] + mag_min[1])/2;  // get average y mag bias in counts
 mag_bias[2]  = (mag_max[2] + mag_min[2])/2;  // get average z mag bias in counts

 dest1[0] = (float) mag_bias[0]*MPU9250mRes*MPU9250magCalibration[0];  // save mag biases in G for main program
 dest1[1] = (float) mag_bias[1]*MPU9250mRes*MPU9250magCalibration[1];   
 dest1[2] = (float) mag_bias[2]*MPU9250mRes*MPU9250magCalibration[2];  

// Get soft iron correction estimate
 mag_scale[0]  = (mag_max[0] - mag_min[0])/2;  // get average x axis max chord length in counts
 mag_scale[1]  = (mag_max[1] - mag_min[1])/2;  // get average y axis max chord length in counts
 mag_scale[2]  = (mag_max[2] - mag_min[2])/2;  // get average z axis max chord length in counts

 float avg_rad = mag_scale[0] + mag_scale[1] + mag_scale[2];
 avg_rad /= 3.0;

 dest2[0] = avg_rad/((float)mag_scale[0]);
 dest2[1] = avg_rad/((float)mag_scale[1]);
 dest2[2] = avg_rad/((float)mag_scale[2]);`

I would like to ask that what is the array dest1 and dest2?
Does this mean that I have to substrate the original mag data with the dest1 for hard-iron calibration? How about dest2?
Thanks in advance!

kriswiner, 2016-08-25:
Take a look here for how these are used:

https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

d9qvwRKks5qjYzugaJpZM4JtBX4> .
<https://github.com/notifications/beacon/AGY1qjGoi1f9WmLXDfqlNUsgaURnzpDgks5
qjYzugaJpZM4JtBX4.gif>

ijunglee, 2016-08-26:
Thank you. I have see that and have other questions of the code. I have posted the questions on the issue.