kriswiner/MPU6050 #15: Comsuption

clemaitre58, 2017-06-07:
Hi,

Do you know the électrical consumption of your STM32F401 and MPU6050 at 5.5kHz ?

Best,
Cédric

kriswiner, 2017-06-07:
No, but for the STM32L4 + MPU9250, which I use now, it is about 3 mA,
deominated by the MPU9250 of course. It is possible to run withe the L4 at
very low power, see:

https://github.com/kriswiner/Ladybug/tree/master/BME280_VEML6040

clemaitre58, 2017-06-08:
Kris,

Could I compile code for the Ladybug board with this chaintools : https://developer.arm.com/open-source/gnu-toolchain/gnu-rm?
How to use JTAG in Debbugging in practice on those board?

Do you also sell MPU9250 breakout board?

Best,
Cédric

clemaitre58, 2017-06-08:
ok I found your MPU9250 board.
If my understanding is great, the M4 in the could be programmed using Arduino IDE?
Does this tools allow also the debug?

kriswiner, 2017-06-08:
Yes, I use the Arduino IDE to program the STM32L4. But Arduino allows no
debugging, yet. The SWDIO and SWDCLK ports are exposed on the dev board I
use so one could connect an ST-LINK and use a traditional tool chain like
Keil or IAR or Crossworks or even gcc/eclipse and program and debug that
way.

clemaitre58, 2017-06-08:
Could you ship today, for France if I place an order?
How long the shipping method take, we need the component for the next week?
Do you have a tuto for Arduino IDE configuration?

kriswiner, 2017-06-08:
Hi Cedric,

Normal shipping via air mail to France will take 8 - 10 business days. You
may also choose Priority International to shorten the time but at
additional expense. If you order in the next few hours I can get it into
the post today.

We do not have an extensive tutorial but we have a relatively mature
Arduino core:

https://github.com/GrumpyOldPizza/arduino-STM32L4

and lots of example sketches:

https://github.com/kriswiner/Butterfly

and, of course, we will help you get started.

Please let's move further discussion to e-mail at tleracorp@gmail.com.

Kris

clemaitre58, 2017-06-08:
Ok, I place the order Now!