kriswiner/MPU6050 #16: Accel Bias

PierreGogendeau, 2017-06-29:
Hi,

I was looking at your file "MPU9250_MS5637_AHRS_t3" and there is something that I don't understand. 
at the line 373 when you used this function :

`accelgyrocalMPU9250(gyroBias, accelBias); // Calibrate gyro and accelerometers, load biases in bias registers` 

Here you store in the internal registers of the MPU9250 biais from the Gyro and Accel. Then line 478 in the main loop you substract it again : 

`// Now we'll calculate the accleration value into actual g's
    ax = (float)MPU9250Data[0]*aRes - accelBias[0];  // get actual g value, this depends on scale being set
    ay = (float)MPU9250Data[1]*aRes - accelBias[1];   
    az = (float)MPU9250Data[2]*aRes - accelBias[2];  `

I thought that when you store the offset value in the MPU9250, it do it by itself to substract the offset. Maybe there is something that I miss understood ? 

Thanks in advance for your help

Cheers

Pierre Gogendeau

kriswiner, 2017-06-29:
I don't like the way the MPU9250 handles the accel bias when written to the
register so I donlt usually write the value there, rather I subtract it
manually in the main code.

PierreGogendeau, 2017-06-29:
Ok thank you for your very very fast answer. Have a good day.

lindig, 2019-02-05:
I believe the bias is corrected by the chip and again in the main loop - and this is wrong. Not doing it in the main loop I can observe that the values are much closer to zero when the chip is not moving. So I suspect a bug here.

kriswiner, 2019-02-05:
The only way the "chip" can correct offset bias is if the user writes the
offset into the bias registers. So yes, double correcting the biases would
produce incorrect results...

lindig, 2019-02-05:
In `MPU6050BasicExample.ino` `mpu.calibrateMPU6050()` is called, which does write the data to the chip, and it subtracts the bias explicitly in the main loop.

kriswiner, 2019-02-05:
Then, of course, this is wrong. These sketches are examples, nbot perfect
black boxes. Buyer beware...