kriswiner/MPU6050 #17: About automatic calibrations of magnetometer

ijunglee, 2017-08-03:
Hello, I have read your article "Simple and Effective Magnetometer Calibration". Do you have any further information about how to do the automatic calibration and let the users know if the magnetometer is well-calibrated and how well they can trust the result after calibration? I can only show the plot figure of mag data to myself; however, I can not know if the magnetometer is well-calibrated and can be used. Thanks in advance!

kriswiner, 2017-08-03:
Once you do the calibration, store the offset biases in your code or EEPROM
and you don;t have to calibrate again.

rogeriojoao, 2017-08-21:
Well defenetly have to buy  you a beer since it is going to save me some time for coding for pic32... there is just one thing that i don't undertand in the for loop shont the FIFO_R_W be updated as ii increments becose if it is not uptadeted we will keep reding the same 12 values over the loop...;)

kriswiner, 2017-08-21:
Not sure what you mean, but in the calbration, the data sampling continues
until the FIFO if full, then the FIFO is read.

rogeriojoao, 2017-08-22:
i forgot tha was a FIFO .... so it puuls up information up as it is read... thanks.