kriswiner/MPU6050 #18: Magnetometer raw data issue

proba238, 2017-08-24:
Hello, in first place I want to thank you for the amazing work about MPU9250. Even dough I’m a student I have very little practical knowledge about sensor fusion (new to the topic), so having a guide like yours means a lot, a ton!

And now about the problem, ill tri to be as simple as possible. So I hooked up my MPU9250, setu up the communication and so on… I can easly access the raw data registers from AK8963. The problem is the measurement. In your article “Simple and Effective Magnetometer Calibration” ,the first picture shows raw data from AKA8963, The blue marks indicates the change in magnetic field along the xy plane. 

I rotated the module for 360 degrees in the xy plane and my measurements of the magnetic field along the plane, with the z pointing to ground shows this:

[image]

(The transition from nord to east is shown by the red rhomb) 

Could you give me an opinion about what the problem might be, and how to fix it?

kriswiner, 2017-08-24:
Can you say what the problem is?

proba238, 2017-08-24:
Hello, and thank you for the quick response!

Well the problem is that I cant use the given magnetic field measurement in the xy to get any reasonable data outcome. If the data would be distributed in a say…circle form as it is the case in :

https://cloud.githubusercontent.com/assets/6698410/8768651/04fe05be-2e3d-11e5-9f70-b6897fdca9ca.jpg

Then I could use a function as tan or tan2 to get the approximate heading where the field is pointing.  Of course I tried to calibrate the sensor as explained in your work but, still. I mean, calibrating just shift the measurement so that the long term run, mean is always at 0,0. 
It seems to me that I should use some form of whited function to calibrate the sensor, a PID or something, in order to scale the measurement to the proper level.

kriswiner, 2017-08-24:
> 11e5-9f70-b6897fdca9ca.jpg

Not sure what to tell you. Try moving the sensor all around in 3D and then
plotting Mx vs My, Mx vs. Mz and My vs Mz, you should get circular clouds
of points.

proba238, 2017-08-24:
All right, ill keep trying till I get them right.  Thanks for the effort! 👍

arkwad, 2017-11-14:
@proba238 
Any progress?

kriswiner, 2018-03-19:
> I have the same problem. I get data from CC2650 to my android phone using
> formula at http://processors.wiki.ti.com/index.php/CC2650_SensorTag_
> User%27s_Guide#Movement_Sensor <http://url>
>
> My decode function:
> [IMG]https://i.imgur.com/6u4jjw8.jpg[/IMG]
>
> Scatter chart:
> [IMG]https://i.imgur.com/1bpwqMZ.jpg[/IMG]
>
> @proba238 <https://github.com/proba238> did you resolve this problem ?

Are you moving the sensor in all 3D directions? Rotating to sample the
entire response surface?

dungtam73, 2018-03-19:
@kriswiner I tried again in all 3D directions. Here is result
[image]
[image]

kriswiner, 2018-03-19:
Well you have a lot of offset bias that needs to be corrected.

dungtam73, 2018-03-19:
Thank you so much! What can I do to get circular clouds of points like you ?

kriswiner, 2018-03-19:
Well, you need to correct the offset biases, rescaling the axes would help
also.

What I do is move the sensor in all directions. Imagine there is an arrow
pointing from the top of the device. During calibration, you want to move
the device such that the tip of the arrow points to every point on an
imaginary sphere surrounding the sensor, multiple times if possible. Slow
and steady. This should give you ellipses. Correcting for offset biases and
axis skew should give you more or less spheres.

See:
https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration