kriswiner/MPU6050 #19: Calibration values question

jc508, 2017-11-03:
Hi,
Thanks for this but I hope you have time for a couple of questions about the code...
I am trying to convert this for use on a QMC5883L chip (sometimes you just have to work with what arrives).
This says its 16bit but does not actually say anywhere what unit-of-measure the registers return, just that they are 16 bit integer from -32768 to 32767.
It also does not have any factory calibration nor scale - or none that are published anyway.

so for the lines of code like ...
dest1[0] = (float) mag_bias[0] * mRes * MPU9250magCalibration[0];
mx = (float)magCount[0]*mRes*magCalibration[0] - magBias[0];
I have eliminated MPU9250magCalibration

In the main program I have also found you have defined mres for 16bit values as.......
 mRes = 10.*4912./32760.0

**Can you please explain the magic of these values** ?

My original sketch had simply
  float heading = (atan2((float)y,(float)x));
  headingDegrees = (heading * C_rad_to_deg); 
this mostly worked except for non-linear values < 50degrees - whereupon people tell me I have to calibrate the animal first !

I think I understand adjusting by the bias ie "- magBias[0]"
and scaling ie mx *= magScale[0];
but the presence of mRes is escaping me.

Thanks in anticipation
JC

kriswiner, 2017-11-03:
mRes just converts the raw sensor values into something more interesting
like milliGauss. 4912 Tesla full scale, 32768 full scale count 10
milliGauss per Tesla, or something like this.
Look at the data sheet for similar:

https://aerocontent.honeywell.com/aero/common/documents/myaerospacecatalog-documents/Defense_Brochures-documents/HMC5883L_3-Axis_Digital_Compass_IC.pdf

palemao, 2021-02-15:
Hi there Kris, first off your Github page is a gold mine. Secondly, in this code 
`// Get hard iron correction
 mag_bias[0]  = (mag_max[0] + mag_min[0])/2;  // get average x mag bias in counts
 mag_bias[1]  = (mag_max[1] + mag_min[1])/2;  // get average y mag bias in counts
 mag_bias[2]  = (mag_max[2] + mag_min[2])/2;  // get average z mag bias in counts

 dest1[0] = (float) mag_bias[0]*MPU9250mRes*MPU9250magCalibration[0];  // save mag biases in G for main program
 dest1[1] = (float) mag_bias[1]*MPU9250mRes*MPU9250magCalibration[1];   
 dest1[2] = (float) mag_bias[2]*MPU9250mRes*MPU9250magCalibration[2];  `

You also use `MPU9250magCalibration`. What is this term for? 

Thanks in advance?

kriswiner, 2021-02-15:
That's the factory calibration of the AK8963C magnetometer. Perhaps poorly
named.

palemao, 2021-02-15:
Is that the same as the Sensitivity adjust values as shown in this picture? 

If not how could I calculate/find this value? What value did you use?

Many thanks 

[image]

kriswiner, 2021-02-15:
Yes.

palemao, 2021-02-19:
Hi Kris, other than accounting  for soft-iron and hard-iron biases, is there any other calculation I need to do before inputting mag data into madwickquaternionupdate() function.  Correcting for magnetic declination perhaps? 

Thanks again

kriswiner, 2021-02-19:
Magnetic declination is accounted for after the filter when you change the
quats into Euler angles. Then True North is AHRS North (i.e., yaw = 0)
minus your declination.

palemao, 2021-02-19:
I just came across this post where you mention accounting for declination in your response. "you have taken into account the
local magnetic declination". https://github.com/kriswiner/MPU9250/issues/345

Just a bit confused now... thanks again

kriswiner, 2021-02-19:
> local magnetic declination". kriswiner/MPU9250#345

a12 =   2.0f * (q[1] * q[2] + q[0] * q[3]);
    a22 =   q[0] * q[0] + q[1] * q[1] - q[2] * q[2] - q[3] * q[3];
    a31 =   2.0f * (q[0] * q[1] + q[2] * q[3]);
    a32 =   2.0f * (q[1] * q[3] - q[0] * q[2]);
    a33 =   q[0] * q[0] - q[1] * q[1] - q[2] * q[2] + q[3] * q[3];
    pitch = -asinf(a32);
    roll  = atan2f(a31, a33);
    yaw   = atan2f(a12, a22);
    pitch *= 180.0f / pi;

*yaw   *= 180.0f / pi;     yaw   += 13.8f; // Declination at Danville,
California is 13 degrees 48 minutes and 47 seconds on 2014-04-04*
    if(yaw < 0) yaw   += 360.0f; // Ensure yaw stays between 0 and 360
    roll  *= 180.0f / pi;
    lin_ax = ax + a31;
    lin_ay = ay + a32;
    lin_az = az - a33;