kriswiner/MPU6050 #20: A question about magnetometer calibration and calibration values.

Lacdires, 2017-11-13:
Hi, I have a  question about magnetometer calibration.
For a simple example, when I rotate magnetometer about Z axis, then I think the magnetometer will measure the same max and min value at X and Y axis, but it get the different values. 
Do you know why?

kriswiner, 2017-11-13:
If the offset biases are different, you will get different resulkts for x
and y, that is the whole point of calibrating the sensor.

Lacdires, 2017-11-13:
Maybe I can ask the question more clear.
The magnetic field will be an ellipse due to the soft iron. Under no hard iron affect, I rotate magnetometer about Z axis, both X and Y orientation will pass the same max and min value, it means 
Xmax - Xmin = Ymax - Ymin
But if they are the same value, it isn't an ellipse.
So I want to know why or something I wrong.

kriswiner, 2017-11-13:
I there are offsets from the origins in the x and y directions (and there
are) then there is no reason to believe xmax - xmin will equal ymax - ymin.
There will be different axial response also, all of this needs to be
calibrated.

Lacdires, 2017-11-13:
Sorry, I cannot understand what you mean.
This is one of my calibration values. (It's can work)
mag_max[0] = 284;
mag_max[1] = 301;
mag_max[2] = 0;
mag_min[0] = -111;
mag_min[1] = -92;
mag_min[2] = -494;
I rotate the magnetometer in a figure eight, so I just want to know when I rotate to let all XYZ's axis positive direction pass the same point, why they didn't get the same value?

kriswiner, 2017-11-13:
The x, y, and z-axes of the magnetometer respond differently. This has to
be calibrated.

Lacdires, 2017-11-13:
OK, so I think it is due to what material to manufacture magnetometer or what principle to let magnetometer work. Do you mean it?

kriswiner, 2017-11-13:
Not sure what you are saying here. Have you seen this:

https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration

It explains more about magnetometer calibration.

Lacdires, 2017-11-13:
Yes, I have seen it. But it is only say hard iron and soft iron will give bias. I want to know if I set my magnetometer at a fixed point and rotate it, it will get the same max and min values or not.

According to my calibration values (I tried to rotate in a small field.), they get the different values, so I want to know the reason.

Alchemy455, 2017-11-14:
Hello, thanks for posting info on calibrating.  I have one question regarding the mag calibration function;
void magcalMPU9250(float * dest1, float * dest2) 
What is the function expecting for dest1 and dest2?  What do these number represent?
Thanks,
-Matt

arkwad, 2017-11-14:
@Alchemy455 
dest1 and dest2 are pointers to arrays from layer that calls magcalMPU9250() method.
In other words : there are output values not input.

Lacdires, 2017-11-16:
OK, I think I got the answer. 
https://www.sensorsmag.com/components/compensating-for-tilt-hard-iron-and-soft-iron-effects
"The distortion produced by soft-iron materials is dependent upon the orientation of the material relative to the sensor and the magnetic field." I think it means the soft-iron distortion affect the material in magnetometer, and different orientation will measure different magnetic field. I set my magnetometer at a fixed point and rotate it, it will get different values at different orientation, so X and Y direction will not get the same max and min values.
Does anyone can answer me what my thought is true or false?

closed 2018-02-02 by Lacdires

kriswiner, 2018-02-02:
The purpose of calibrating the magnetometer is to remove offset biases
(hard iron) and axial distortions (soft iron) effects. Usually, for good
enough results, simple recentering the response surface on the origin and
scaling the x/y/z axial responses to correct axial response mixmatches is
enough. This is what is done in most of my sketches. This amounts to
essentially the diagonal terms in a 3 x 3 matrix. Once can gof further and
use the full 3 x 3 matrix apprach to correct the response surface back to
aperfect sphere, which will of course give best results.

In all cases, no calibration will result in wrong heading with large cross
axis effects, unless just by chance the magnetometer offset biases and
axial corrections are small as built, which is very unlikely.

So the bottom line is, calibrate your magnetometer  for at least the hard
iron effects and you should see greatly diminished artifacts as you
describe.