kriswiner/MPU6050 #23: Strange reset behaviour

Yury-MonZon, 2018-01-26:
It works as intended after the first sketch upload. After that if I unplug usb cable thus cutting the both devices power and plug back - the fun starts here) The sketch is working, but this time the response from mpu9250 is like 10 times slower. Here is the weird part - it still works, but just shows the data slower in Serial Monitor and the led blinks slower. 

What am I doing wrong? I'm using Arduino Pro Micro (324u4 chip) and 9250 board. Thanks.

kriswiner, 2018-01-26:
No idea, but generally micrcontrollers only do what you tell them to...I
doubt this is an MPU9250 problem.

Yury-MonZon, 2018-01-29:
The fix found - wrong bootloader was flashed in the Arduino Pro Micro chip. Had to reflash it using AVR ISP mk2 and Arduino IDE. So if you got something like and your board is not genuine SparkFun Pro Micro - then reflash the bootloader.

closed 2018-01-29 by Yury-MonZon