kriswiner/MPU6050 #24: pitch and roll variation

hassankord, 2018-03-01:
Dear Kris,
many thanks for the great code you provided, 
I used arduino nano to collect data from the mpu-6050 sensor using the code 
/* MPU6050 Basic Example with IMU  
 by: Kris Winer
 date: May 10, 2014
I removed the display and added a serial print to get the result. 
I collected the data for more than 10 minutes and got the following summarized result 

ROLL : minimum   -0.31 ;  maximum 1.00 ;   ravage   0.322729 ;  variation +- 0.65
Pitch : minimum   -0.02 ;  maximum 1.30 ;   ravage   0.64 ;  variation +- 0.65

as you can see the variation is high- around one degree - while in the DMP its around 0.1 degree for roll and pitch.

do you get the same result or have I done a mistake?
your clarification is highly appreciated.

Best Regards,
Hassan
 
[image]

kriswiner, 2018-03-01:
I haven't looked at this for a couple of years. First of all, 1 degree
error in pitch and rioll is pretty good. The DMP heavily filters this
result and you can to, just use a moving average and you can cut the
variation by half easily.

hassankord, 2018-03-02:
many thanks

closed 2018-03-02 by hassankord