kriswiner/MPU6050 #30: InvenSense ICM-20948 DMP Flash error

qasimz, 2018-03-16:
Hi,
I am using ICM-20948 by TDK InvenSense, which has on board digital motion processing unit. My host processor is MediaTek MT-2523G.
InvenSense provides DMP firmware, that has to be programed on DMP to access it features. I am using I2C to communicate with ICM-20948. I am able to get the RAW accel data. But when I try to write the firmware bytes on DMP, somehow my writes are not getting to its memory. 
I've seen on InvenSense developers form, most of the people are facing this issue. When I write the bytes to DMP memory, I get ACK from the device, but when I try to verify the DMP memory, I recieve garbage values. 
I would be very thankful, if I get solution of this problem.

Furthermore, is there any activity recognition library, which take axis data as input and tells us user's activity.

kriswiner, 2018-03-16:
Sounds like a question for Invensense.

We use the EM7180+MPU9250 to identify sitting, standing, walking, etc. The
algorithms have to be developed by the user though, we can't share ours.But
it is straightforward.

tatulea, 2018-09-05:
@qasimz can you send me an email, please? I am interested in finding more about ICM-20948 (arnautug7@gmail.com)

Thanks

dtquang89, 2019-04-09:
Hi, do you have any solutions for this issue? Thank you.

kriswiner, 2019-04-09:
No, try the Invensense Developer's forum.

qasimz, 2019-04-10:
> Hi, do you have any solutions for this issue? Thank you.

which communication protocol are you using?

dtquang89, 2019-04-10:
I’m using SPI.

qasimz, 2019-04-10:
In ICM20649Transport.h  what values have you set for INV_MAX_SERIAL_READ and INV_MAX_SERIAL_WRITE?

I have set them to 1. 

I am also posting my spi send and recv routine.

`static int spi_master_transfer_tx(void * context, uint8_t register_addr, const uint8_t * value, uint32_t len)
{
  
	
	data[0] = register_addr & WRITE_BIT_MASK;
	for ( int i = 0 ; i < len ; i++)
	{
		data[i + 1] = *value;
		value++;
	}
	
	spi_send_and_receive_config.send_length = 1;
	spi_send_and_receive_config.receive_length = len + spi_send_and_receive_config.send_length;
	spi_send_and_receive_config.send_data = data;
	spi_send_and_receive_config.receive_buffer = status_receive;
	if(HAL_SPI_MASTER_STATUS_OK != hal_spi_master_send_and_receive_dma_blocking(ICM_SPI_CHANNEL,&spi_send_and_receive_config))
		return -1;
	return 0;
	

}

static int spi_master_transfer_rx(void * context, uint8_t register_addr, uint8_t * value, uint32_t len)
{
	//hal_gpt_delay_ms(200);
	spi_send_and_receive_config.send_length = 1;
	spi_send_and_receive_config.receive_length = len + spi_send_and_receive_config.send_length;
	data[0] = register_addr | READ_BIT_MASK;
	spi_send_and_receive_config.send_data = data;
	spi_send_and_receive_config.receive_buffer = status_receive;
	if(HAL_SPI_MASTER_STATUS_OK != hal_spi_master_send_and_receive_dma_blocking(ICM_SPI_CHANNEL,&spi_send_and_receive_config))
		return -1;
	//printf("\r\n %x \r\n",status_receive[1]);
	for (int i = 0 ; i < len ; i++)
	{
		*value = status_receive[i+1];
		value++;
		//printf("\r\n status = 0x%02x \r\n",status_receive[i+1]);
	}
	return 0;
	
}`

dtquang89, 2019-04-16:
Thank you for helping me. Actually changing the values doesn't solve the problem. I am using ICM with nRF52. I'm now sure why.

qasimz, 2019-04-16:
Which solution worked out for you? 

---- On Tue, 16 Apr 2019 20:57:20 +0500 notifications@github.com wrote ----

Thank you for helping me. Actually changing the values doesn't solve the problem. I am using ICM with nRF52. I'm now sure why.

dtquang89, 2019-04-16:
I haven't found a solution yet. I'm still looking for it.