kriswiner/MPU6050 #32: Sensor Temperature Compensation

StarCountingCode, 2018-04-15:
Hello there! First off, huge thanks for this library, it has saved me lots of trouble already.
However, there is one thing I do not quite know how to do, and this is gyroscope/accelerometer temperature compensation. I have experienced significant drift when the sensor is not freshly calibrated, and would like to manually change the offsets with varying temperature. Is there an equation or something I am missing? Or is the only way to gather data in different temperature conditions?

kriswiner, 2018-04-15:
I am afraid you will have to calibrate this, since each board will be
slightly different. The good news is the behavior should be quadratic in
tenperature like any crystal oscillator, so it should be failry easy to
come up with a calibration table for each accel axis. The gyro will be
harder...

StarCountingCode, 2018-04-15:
I see. Looks like there is no way around that. Thank you for the advice, I'll see what I can do.

closed 2018-04-15 by StarCountingCode