kriswiner/MPU6050 #34: seek help：how can i get the angle without using Arduino

mengjiehehe, 2018-07-02:
Hello , sir . in your demonstration，we can get information from mpu6050 by using arduino,but if  i want get   information by mpu6050 and usb-i2c transverter instead of arduino ，what should a freshman do .Can you give me a train of thought，Thanks a million！

kriswiner, 2018-07-02:
Maybe you can use the DMP, but generally you need some device to issue I2C
commands. What would it be in this case, the pc or linux box?

mengjiehehe, 2018-07-06:
Thans for your reply, i want to use the dates of mpu6050 in pc of  linux  system and i get the information by CUTECOM. But so far i have not found a driver that I can use directly. Maybe I should learn more about it ,and then write my own driver for it .and your demo is good example for me,thanks a lot .

kriswiner, 2018-07-06:
Better is to use an MCU to collect the data from the MPU6050, and send the
data to the pc via USB serial.

Tednsn, 2018-07-10:
Thanks a lot.

closed 2018-07-10 by Tednsn