kriswiner/MPU6050 #36: Pb with self-test

jmcornil, 2018-09-10:
Hello Kris

First of all, thank you for your work and for sharing it.

I am a beginner with MPU6050 and I am reading your code in order to understand the lacks in the documentation of Invense. I have two questions
1. In your SelfTest() function code, there are lines such as : 
    selfTest[0] = (rawData[0] >> 3) | (rawData[3] & 0x30) >> 4
   I wonder if  (rawData[0] >> 3) would not be replaced with (rawData[0] >> 3) & ~0b11
   in order to kill bit-0 and bit-1 ?
  Without & ~0b11, it could generate a little error up to 3, no ?
2. I  have also a question about bit-0 in accel bias register.
   Can we arbitrarily set them to 0 or 1, and what is the influence of each choice 
   on the temperature compensation? 

Thank you in advance for your reply

J.M. CORNIL

jmcornil, 2018-09-12:
About the (2.) question
Yes, I saw it on your code but at the beginning, I executed a script by jF ROWBERG
https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/examples/IMU_Zero/IMU_Zero.ino
and it seems not to me that he takes account of these bit-0.

Is it possible to retrieve the initial value that Invensense wrote in the register ?

Where can I find info about these bit ?
For the only one I found is on the page 7 the MPU Hardware Offset Registers Application Note
It is this line
Bit 0 on the low byte of each axis is a reserved bit and needs to be preserved
which caught my attention and encourage me to search on Google until I reach your work.

kriswiner, 2018-09-12:
That is the only source I know.