kriswiner/MPU6050 #37: magnetic calibration

robotdreamers, 2018-10-12:
Hi Kris, for 8-shape magnetic calibration. Do i need to move in 8-shape for all three dimensions (x, y, z)?
This what happens when I did on all directions on my chip. I am using uT units. And does it has to be around 0 like what you have done. Thank you very much.

[image]

kriswiner, 2018-10-12:
Yes all three directions, really over all possible directions over the
entire 3D response surface.

robotdreamers, 2018-10-12:
I my location it looks like this, does it mean mZ = 480mG. what can I tell about mX and mY. Thanks.

[image]

kriswiner, 2018-10-12:
It means SQRT(Mx**2 + My**2 + Mz**2) ~ 480 mG, as expected. Look at the
USGS for local field in your area, probably not too different from
Danville, California.

robotdreamers, 2018-10-12:
so how should I read this :). so many jargons for an amateur :(

[image]

kriswiner, 2018-10-12:
Huh?
227 mG, 54 mG, 420 mG is this what your mag shows?

robotdreamers, 2018-10-12:
No i got from this website for my location: https://www.ngdc.noaa.gov/geomag-web/#igrfwmm

robotdreamers, 2018-10-12:
ah so I am supposed I should expect: mX to be 227mG, mY 54mG and mZ 419mG.