kriswiner/MPU6050 #39: FIFO_R_W

KannanNatesh, 2019-01-10:
Dear Friends,
i have working with MPU6050 and (I2C) 8byte register Microcontroller,
i have problem in adding this line
readBytes(MPU6050_ADDRESS, FIFO_R_W, 12, &data[0]); // read data for averaging

because this line shows to get 12 byte from the registers
but my microcontroller have only 8byte register,

so can i write the above line like
readBytes(MPU6050_ADDRESS, FIFO_R_W, 6, &data[0]); 
    accel_temp[0] = (int16_t) (((int16_t)data[0] << 8) | data[1]  ) ;  
    accel_temp[1] = (int16_t) (((int16_t)data[2] << 8) | data[3]  ) ;
    accel_temp[2] = (int16_t) (((int16_t)data[4] << 8) | data[5]  ) ;
readBytes(MPU6050_ADDRESS, FIFO_R_W, 6, &data[0]);     
    gyro_temp[0]  = (int16_t) (((int16_t)data[0] << 8) | data[1]  ) ;  
    gyro_temp[1]  = (int16_t) (((int16_t)data[2] << 8) | data[3]  ) ;
    gyro_temp[2]  = (int16_t) (((int16_t)data[4] << 8) | data[5]  ) ;

i hope i get answer soon as possible thank you,

with regards 
kannannatesh.S

KannanNatesh, 2019-01-10:
Dear Friends,
for your reference i add some code what i write for get raw value from accelerometer registers.
i have read accelero meter value with this code
u8 accreg[6]={0x3B,0x3C,0x3D,0x3E,0x3F,0x40,0x00};
u8 read_buffer[8]={0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00};
ret = Ql_IIC_Write_Read(1, addr, accreg, 1,read_buffer, 6);	//1-is Channel,   addr=0xd0,    accreg-is accelero meter registers       1=must be true for communication enable,     read_buffer-is read data from slave registers,        6= registers length;
	dec_accx=(read_buffer[0]<<8) | read_buffer[1];
	dec_accy=(read_buffer[2]<<8) | read_buffer[3];
	dec_accz=(read_buffer[4]<<8) | read_buffer[5];
	SerialPrint("--AX%d",dec_accx);
	SerialPrint("--AY%d",dec_accy);
	SerialPrint("--AZ%d",dec_accz);

thank you 

with regards
kannannatesh.S

kriswiner, 2019-01-10:
Yes, you can break up the read/write into 6 or 8 byte chunks. Is this your
question?

KannanNatesh, 2019-01-11:
Dear kriswiner,
ya that is my question, my controller connect 0xd0 address with Accelerometer, so 0x75 register returns 0x58, my controller read data in only 8bytes in single transmission,

and one thing i am newbie for I2C Communication so i don't know to how to read/write data to the registers  

the below code shows my controller how to read data from registers
must i write address for each transmission like
  u8 accreg[6]={0x3B,0x3C,0x3D,0x3E,0x3F,0x40,0x00};
this above char store 6 byte address for read 6 registers,
 ret = Ql_IIC_Write_Read(1, addr, accreg, 1,read_buffer, 6); //1-is Channel, addr=0xd0, accreg-is accelero meter registers 1=must be true for communication enable, read_buffer-is read data from slave registers, 6= registers length;

my doubt is 
if i read single byte data from FIFO 0x74 register that data is still available in same register or that is  Automatically erased,
because i send request to read 2nd byte data from same address 0x74 register that is read same data(like first request read data) or not. 

sorry for my non grammatical English
 I hope you understand my doubt/question

thank you

with regards 
kannannatesh

kriswiner, 2019-01-11:
I have no idea how to use your controller. The FIFO is designed to be burst
read, but there is no reason you can't read it 8 bytes at a time.

These problems seem to have less to do with the MPU9250 sensor than the
specific weaknesses of your MCU. Why not get a real MCU, like this
<https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/>
one?

KannanNatesh, 2019-01-11:
Dear kriswiner,
thanks for your suggestion that is ok for my project but already board was designed with MPU6050 so i can't change the module,

if you can please help to clear my doubt 
if i read first byte from 0x74 FIFO register that byte is still available in FIFO register or that is automatically erased.

thank you'

with regards
kannannatesh

kriswiner, 2019-01-11:
No idea, check the data sheet.

KannanNatesh, 2019-01-12:
k thank you