kriswiner/MPU6050 #44: How to produce the Mxy Mxz Myz point picture?

FN0005, 2019-10-30:
Is it generated using matlab's plot3 or scatter3 code and then converted to 2D picture?
Can provide a reference code?Thanks!

[image]

kriswiner, 2019-10-30:
Output Mx, My Mz on the serial monitor and use excel or equivalent to plot
one vs the other.

FN0005, 2019-10-30:
Use excel scatter plot function？
Mxy is z vs xy?

kriswiner, 2019-10-30:
Mx vs My, for example, any plot program will do, really not that hard....

FN0005, 2019-10-30:
Ok,for example,the attached file is my magnetometer data but I don't know how to use the excel scatter plot function because we need to project the XY YZ XZ generation point onto the plane,so do you have any reference excel file for reference?Thanks!
[01.xlsx](https://github.com/kriswiner/MPU6050/files/3787218/01.xlsx)

kriswiner, 2019-10-30:
I have no idea what you mean. I simply plot Mx vs My, Mx vs Mz, etc. Why do
you need to do more than this?

FN0005, 2019-10-30:
I understand what you mean but how do you place the Mxz Mxz Mxz in the same plane?

FN0005, 2019-10-30:
This is my question
[image]

closed 2019-10-30 by FN0005