kriswiner/MPU6050 #47: Calibration Function

NadaSaad-D8, 2020-10-07:
hello Kiris, thank you so much for your great work i am new to MPU and it is really helpful , i have a question i cannot understand the algorithm of the calibration function , could you send me any paper that talks about calibration you followed in your code ?

kriswiner, 2020-10-07:
The calibration simply collects data when the sensor is motionless,
calculates the average and then subtracts the result from subsequent data.
This has the effect of centering the response surface at the origin. No
paper is required to understand this basic function. But a google search
should find many discussions of this.

NadaSaad-D8, 2020-10-07:
thanks for your reply , 
so i have many questions if you have time i will be thankfull if you explained this , i searched alot and the calibration functions are different from yours whatever i don't fully understand them , i got that the idea that i calculate many samples and got the average then subtract it from from subsequent data. but when i follow the code i didn't find it that way 
FIRST : what are you reading here ? , why don't we say i read for example 1000 sample so i will divide by 1000 after reading them
 readBytes(MPU6050_ADDRESS, FIFO_COUNTH, 2, &data[0]); // read FIFO sample count
  fifo_count = ((uint16_t)data[0] << 8) | data[1];
  packet_count = fifo_count/12;// How many sets of full gyro and accelerometer data for averaging
SECOND : 
Next step is simple we are reading the data (Packet count number) then divided them by packet count to get average 
till we reach this , i didn't get what is the meaning of removing sensor senstivity, how is this related to gravity 
and then dividing by 4 , and in gyro dividing by 8
 if(accel_bias[2] > 0L) {accel_bias[2] -= (int32_t) accelsensitivity;}  // Remove gravity from the z-axis accelerometer bias calculation
  else {accel_bias[2] += (int32_t) accelsensitivity;}
 
// Construct the gyro biases for push to the hardware gyro bias registers, which are reset to zero upon device startup
  data[0] = (-gyro_bias[0]/4  >> 8) & 0xFF; // Divide by 4 to get 32.9 LSB per deg/s to conform to expected bias input format
  data[1] = (-gyro_bias[0]/4)       & 0xFF; // Biases are additive, so change sign on calculated average gyro biases
  data[2] = (-gyro_bias[1]/4  >> 8) & 0xFF;
  data[3] = (-gyro_bias[1]/4)       & 0xFF;
  data[4] = (-gyro_bias[2]/4  >> 8) & 0xFF;
  data[5] = (-gyro_bias[2]/4)       & 0xFF;
 
i am so sorry if the questions are trivial , thanks in advance

kriswiner, 2020-10-07:
I am using the FIFO buffer to collect the samples, The approach is the same.

palemao, 2021-06-08:
Hi Kris, your simple method for calibration is very effective, however I wanted to implement an even more accurate calibration method that are suited for commercial applications of autonomous vehicles. Are there any you would recommend?

Thanks in advance!

kriswiner, 2021-06-08:
> Here <https://github.com/gregtomasch/EM7180_SENtral_Calibration> are some
> calibration comments specifically for the EM7180 boards but generally
> applicable for accel calibration.
>
> For mag calibration, there is the Freescale solver or equivalent,  which
> should be available as an open-source library. for example this

See also this
<https://community.nxp.com/t5/Sensors/Magnetometer-calibration-is-not-always-accurate/m-p/410482>.
And keep googling...