kriswiner/MPU6050 #48: offset increasing

yousefjoe175, 2020-10-09:
Hello mr kris,Thank you for your great effort and dedication. Your code has really helped me a lot.
I tested your code using MPU6050 sensor and an arduino uno board it works fine while i keep rotating the sensor but when I stop moving the sensor the values of yaw, pitch and roll angles increase by time. So the offset increases and affects the future outputs.
Please, do you have any idea what am i doing wrong?
Thanks in advance

kriswiner, 2020-10-09:
Well, the MPU6050 can't really be used for heading estimation because there
is no magnetometer to compensate for gyro drift. So I would expect you
should get good estimation of pitch and roll (if your sensor is properly
calibrated) but heading will drift at ~several degrees per minute or more).

yousefjoe175, 2020-10-09:
But we can't use the magnetometer due to the frequent change in the magnetic field in the environment we will put the sensor in. Beside all we need is just yaw. So, do you have any suggestions about how we can solve this issue?

kriswiner, 2020-10-09:
Can't get accurate yaw without a magnetometer, at least not for more than a
few seconds. Try calibrating your sensors for at least good enough pitch
and roll.

We use 9 DoF solutions (i.e., with a magnetometer( even in harsh magnetic
environments like small drone with four brushed motors. Works just fine.
Maybe you need something like this
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-mpu9250/>
or maybe even this
<https://www.tindie.com/products/onehorse/max32660-motion-co-processor/>.