kriswiner/MPU6050 #50: Madgwick Quaternions without accelerometer data

intensite, 2020-10-30:
Hello Kris,

First of all, thanks for this old but very useful library.

Is it possible to use the MadgwickQuaternionUpdate without Accel inputs?  

I am in the process of creating a model rocket flight computer and I suspect the fusion of the Gyro with accelerometer cannot be used in this application because of the high acceleration inherent to rockets.

I do however need the integration of the angular velocity of the gyro axis into Quaternions.

I tried to strategically place a if statement in the code of the MadgwickQuaternionUpdate to ignore the Accel data if 0,0,0 is passed for the 3 accel axis.  However, I am unfortunately not gifted for complex algebra and so far I haven't had much success. 

Is this something you can help with?

Best regards.

kriswiner, 2020-10-30:
" Is it possible to use the MadgwickQuaternionUpdate without Accel inputs?"

As far as I know the answer is no.

What you can do is put a small weighting on the accel values, sort of like
setting them all to zero, but I am not sure this will work too well. Maybe
have a weighting of 1 if the absolute value of the accel vector is 1 g or
less and an exponentially decreasing weighting as the accel increases above
this value?

intensite, 2020-11-01:
Thank you sir!

One would alter these using the beta or the zeta parameter?
Right now (by default) my beta evaluates to 0.60 while zeta evaluates to 0.03

kriswiner, 2020-11-01:
These parameters are really irrelevant...but you can try different values
if you want.

Most important are proper calibration of the sensors, and fast enough speed
(number of iterations per new data sample) of the fusion rate. Everything
else is in the noise...