kriswiner/MPU6050 #55: Good job. Is your libary maintained yet?

f41ardu, 2022-01-19:
[image]

Works very well. Is work in progress expected? 
And what is the best way to get start values for gyroMeasError and GyroMeasDrift.

kriswiner, 2022-01-19:
What kind of work in progress?

f41ardu, 2022-01-24:
For example improvement and maintenance. Just to come back to my question. How do you estimate initial values for are beta and zeta values for quaternion calculation. And at least the changes after stabilisation?

kriswiner, 2022-01-24:
I usually ignore these values. Use whatever is in the sketches. The
results do not depend strongly on these parameters. Results depend on 1)
quality of sensors, 20 calibration of sensors, 3)  rate of fusion (speed of
MCU) in that order.