kriswiner/MPU6050 #56: MPU6050 Calibrated Acc and Gyro in different measurement range

williamesp2015, 2022-01-28:
If you see source code for IMU you will see using DMP or raw data but none has given more realistic solution for applied project. I'm using raw data and I get offset when MPU6050 is set to measure +/-2g and 250 DPS. I store them in an int16 array so aRes is Acc rang and gyro rang /32768.0. Now, when I read sensor in a loop, I have to get calibrated value:
(float) (Ax-(offsetAX/ADIVIDER))*aRes;
(float) (Gx-(offsetGX/GDIVIDER))*gRes;
ADIVIDER for 2g is 1 and for 16g is 8
GDIVIDE  for 250 DPS is 1 and for 2000 DPS is 8
Even though, the method is much better than others but in gyro I see offset about 1-7 as I expected to have 0.

bhupiister, 2022-01-28:
Search for a library i2c_mpu6886_4khz by yururi. Mpu6886 and 6050 are identical and this library is easy to use for starters.

kriswiner, 2022-01-28:
I would say that you should perform the calibration for each full scale
setting you want to use. Not sure why you would want to change this
dynamically, but if you do, you will need to calibrate the offset bias
again.