kriswiner/MPU6050 #57: does calibrateMPU6050 method push biases to hardware register?

audi0615, 2023-02-23:
Hi.

I have a question regarding calibration method.

In MPU6050BasicExample.ino, I found that calibrateMPU6050 is done in setup() and the biases are subtracted to accel & gyro data again in loop().

does this mean the calibrateMPU6050 method only calculate the biases without pushing them to hardware register?

kriswiner, 2023-02-23:
yes..you can push them to registers but I found this to be more trouble
than it was worth...

audi0615, 2023-02-23:
Thanks Mr. Winer

But do you mean that it is supported in MPU6050 but not recommended? The calibrateMPU6050 method is pushing them to registers as shown below. If this works, then it seems there is no need to subtract them from accel and gyro data.

`  // Push gyro biases to hardware registers
  writeByte(MPU6050_ADDRESS, XG_OFFS_USRH, data[0]);// might not be supported in MPU6050
  writeByte(MPU6050_ADDRESS, XG_OFFS_USRL, data[1]);
  writeByte(MPU6050_ADDRESS, YG_OFFS_USRH, data[2]);
  writeByte(MPU6050_ADDRESS, YG_OFFS_USRL, data[3]);
  writeByte(MPU6050_ADDRESS, ZG_OFFS_USRH, data[4]);
  writeByte(MPU6050_ADDRESS, ZG_OFFS_USRL, data[5]);

  // Push accelerometer biases to hardware registers
  writeByte(MPU6050_ADDRESS, XA_OFFSET_H, data[0]); // might not be supported in MPU6050
  writeByte(MPU6050_ADDRESS, XA_OFFSET_L_TC, data[1]);
  writeByte(MPU6050_ADDRESS, YA_OFFSET_H, data[2]);
  writeByte(MPU6050_ADDRESS, YA_OFFSET_L_TC, data[3]);
  writeByte(MPU6050_ADDRESS, ZA_OFFSET_H, data[4]);
  writeByte(MPU6050_ADDRESS, ZA_OFFSET_L_TC, data[5]);`

kriswiner, 2023-02-23:
Not sure this is supported for the MPU6050. I recommend you look in the
data sheet and verify.

You can also try both methods (in line and via register writes) and
verify which works.

You can also read the offset registers to verify that the calibration data
is properly stored there after calibration.

What I am saying is that I did all of this years ago and chose to correct
for calibration bias in the main loop. YMMV.

audi0615, 2023-02-24:
Thanks Mr. Winer.

I could find a document which confirms the MPU6050 has the hardware offset registers.
https://www.digikey.com/en/pdf/i/invensense/mpu-hardware-offset-registers

I would verify which works, thanks again.