kriswiner/MPU6050 #58: Direction for the compass inside Ublox M8N Gps.

saumik02, 2024-06-19:
which direction should i mount the GPS . What is the relation to the axis of mpu6050?

saumik02, 2024-06-19:
The GPS has a inbuilt HMC5883l compass in it
[image]

saumik02, 2024-06-19:
[image]

kriswiner, 2024-06-19:
No idea about the M8N, best ask UBLOX

saumik02, 2024-06-26:
I have solved that issue. But I am getting the Euler angle. Is it right? Then what are the formulae for converting it to a direction vector? By the way, I am making a filter using UKF using your quaternion.

saumik02, 2024-06-26:
I am using the NED frame as a reference.  I am giving you some ss of my results. Please, guide me to the right path. From left yaw>roll>pitch.
When the X-axis is pointing north.
[image]

when the X-axis is pointing West.
[image]

when the X-axis is pointing South.
[image]

when the X-axis pointing east.
[image]

kriswiner, 2024-06-26:
This is not looking correct to me. If the sensor is laying flat on a table,
the pitch and roll shouldn't change as you rotate the sensor but the yaw or
heading should change by ~90 degrees when the sensor is rotated by 90
degrees, etc.

My guess is that the sensors (accel, gyro, and mag) are not properly
calibrated.

saumik02, 2024-06-26:
Can I have a picture of your setup for this filter? It may be helpful for me to understand my sensor orientation is right or wrong.

kriswiner, 2024-06-26:
I haven't used the MPU6050 for more than five years and I am not using the
MAX M8N so I am not able to help you with a picture of my "setup".

I would recommend you check the basic sensor output data.

At rest and flat on a table, the accel should output 0, 0, 1000 milligs,
the gyro should output 0, 0, 0 millidps, and the mag should output whatever
is expected for your location (for me in California I expect 0.2, 0.1, and
0.4 Gauss), where the total magnetic field should be ~0.5 Gauss. You can
look up what to expect for the local magnetic field using the government
geological website or some such. Errors on these should be ~1% or less

Anyway, until you can get these basic sensor results, the fusion filter and
Euler angles are irrelevant.